TomWhitwell/TuringMachine #195: No Noise nor randomness

MFallise, 2023-02-10:
Good day,

First tests on my newly built Turing looked fine until I discovered that actually, randomness was missing. After double checking soldering, re-calibrated and some additional tests, here is my status:
* Sequencing is fine (length selection behaves correctly, clock ok)
* Scale => OK
* CV OUT => OK
* PULSE => OK
* Lock trimmer => looping Ok (see below)
But...
* Nothing comes out of Noise OUT
* Randomness (12 o'clock on trimmer) keeps looping on the same sequence without any variation
So...
I've could create a "randomness" by patching a LFO in the CV IN, then, trimmer creates some "fake" randomness (actually following the LFO) and can lock (full CW ) on a sequence. But that's not what the baby is made for, right...
This looks like close to this issue: https://github.com/TomWhitwell/TuringMachine/issues/97
Based on this thread, I've checked the schematics and this really looks like local the the noise generator circuit. As I see, there could be only 2 cause for my problem: either the 2N3904 or the TL074... IMO, assuming that I get a correct CV OUT which also uses the TL, it seems most unlikely that it's damaged (am I correct ?). So, all my suspicions focuses on the 3904...
Could you suggest a workaround or a way to verify my assertion ? 

Cheers,

Mike

TomWhitwell, 2023-02-10:
Hi, sorry to hear that, no noise will mean no randomness so that sounds right

MFallise, 2023-02-10:
Thank you Tom for this quick reply. Do you think of any advice to troubleshoot/ locate the issue ?

MFallise, 2023-02-17:
Hi Tom, ordered a new 3904, and happy to say that everything works like a charm since then. Meanwhile I ordered the Voltages expansion and, boy, this is some nice piece of hardware!
So this issue was relying on a overcooked transistor and is now closed for me. A hint for amateur electricians like me: Be very careful when soldering transistors, they are very sensitive to high temperatures.

Cheers, 

Mike

closed 2023-02-17 by MFallise