TomWhitwell/TuringMachine #197: All lights on at power-up, output inverted?

mightysmooth, 2023-02-26:
I just completed my first TM mkII but it doesn't work as expected. When I power up all the 8 lights at the top of the module stay on even with no clock present. I have connected up to the pulse output of an oscillator (as per the calibration video) and I get some random output but the signal seems inverted, with the range pot fully CW the module outputs a constant ~10V which then drops towards zero on the steps. Locking CW causes the output to become intermittent, I seem to get some loop but it changes every few seconds and eventually goes back to a steady ~10V output. I have a Volts expander and the output of that exhibits the same behaviour. I have tried disconnecting the Volts expander but it makes no difference. All ICs have been reseated and I have checked all my solder joints (the soldering probably isn't perfect!). Images of the boards attached.
[image]
[image]
[image]
[image]

mightysmooth, 2023-02-26:
A few other things I've checked:
- Noise output seems fine ~8V peak-to-peak
- Output voltage changes from 0 to ~+10V with the range pot as expected
- I've tried a few different clock sources/oscs and also tried changing the pulse width but there is no noticable effect
- With the top board disconnected the shift registers all output ~+12V on every output pin

TomWhitwell, 2023-02-26:
Sorry to hear you're having trouble. 

**When I power up all the 8 lights at the top of the module stay on even with no clock present.** 
This is normal behaviour 

**I have connected up to the pulse output of an oscillator (as per the calibration video) and I get some random output but the signal seems inverted, with the range pot fully CW the module outputs a constant ~10V which then drops towards zero on the steps.** 
Do you mean when LEDs are off, you get max voltage out? That's very much not normal behaviour 

 **I have checked all my solder joints (the soldering probably isn't perfect!). Images of the boards attached.**
Soldering looks good to me 

**With the top board disconnected the shift registers all output ~+12V on every output pin**
That might not be odd if there is lots of noise going through the shift regs it could appear that they are all on all the time

mightysmooth, 2023-02-27:
Now that I have done some more testing I think the problem may lie with the DA conversion somewhere. Taking account of the fact I haven't been able to calibrate yet, the behaviour of the LEDs seems to align pretty well with what I'd expect in terms of randomness and locking with the main knob positions. Its just that the output voltage doesn't seem to reflect what I'm seeing with the LEDs! It was too difficult to spot this when I was using the oscillator to clock it for the calibration procedure.
I presume the input to the DAC must be ok as the LEDs are reflecting the right sequence but I don't have suitable equipment to check this and confirm. Am I on the right track here do you think?

mightysmooth, 2023-03-01:
I've checked the circuit with an oscilloscope and it looks like the DAC is getting all the right input. I have checked the DAC output and it is always negative, is this expected? The symptoms sound very similar to those reported in issue #178 but I don't know if that was ever resolved? Voltage at pin 14 of the DAC is 0V when the DAC is in place but changes to around 2.5V with the DAC removed. Is this indicative of a broken DAC do you think? If so, do you have any suggestions for a replacement as I'm struggling to find another DAC0800LCN in stock online.

TomWhitwell, 2023-03-02:
Yes, that all makes sense. You can check the TL074 on the front board - Pins 1,2,3 are the circuit that scales the DAC output - if the pin 2 or pin 3 connections is bad you might have these symptoms 
If it is the dac - in the UK, Bitsbox have DAC0800 in stock, otherwise try support@thonk.co.uk ?

mightysmooth, 2023-03-12:
Problem solved, I focussed my attention on the circuit around the TLO74 and there was a build error. I had managed to swap the 5.1K and 51K resistors for R26 and R5, having corrected that mistake the module now works flawlessly. Many thanks for the assistance Tom

closed 2023-03-12 by mightysmooth