TomWhitwell/TuringMachine #198: Pulse Not Working

drcat67, 2023-03-16:
Hi,
Just finished building the Turing Machine.
I successfully calibrated the out (Although the screw inside the trimmer totally disintegrated)
The pulse does not output anything, plus its led does not light up.
At this point, I am not sure if the machine is functioning as it should. I mean how do you know if it emits the right voltage when everything is random?

TomWhitwell, 2023-03-16:
Hi, 
Sorry to hear that. 
If the voltages are changing when the LEDs changing, and they're roughly in the range 0v to 8v when scale is fully up, then that output is working. 
If the pulse isn't working, it might be a missed solder connection - on [the schematic](https://www.musicthing.co.uk/collateral/TuringMachine_schematic.pdf) I can see that Pulse out comes from one of the the 4050 chips on the back board that drive expanders. If you don't have an expander connected, you might not notice a problem there - so start by checking that the 4050 chips are the right way round, seated properly, with all the pins soldered.

drcat67, 2023-03-16:
HI Tom,
Thanks for your reply. I did buy pulses too which I have now built but I only get 5 6 7 working (Led Flashing).
I looked at these 2 chips I cannot really see much wrong with them.
So I have 3 out of 12 in the pulses module and none in the Turing.

TomWhitwell, 2023-03-16:
OK so if the LEDs on the main module are all working, but pulses isn't, the problem may be the 4081 chips on the back - maybe one is the wrong way round or there are missing solder joints. Possibly the chip marked AND_1 on the schematic.

drcat67, 2023-03-17:
Hi Tom,
What is the best way of testing the ICs with a multimeter? Sorry the question may sound silly, but I am learning here by crash landing, not ideal, but I do not have a clue and what I found online so far is a lot of fluff and very few practical tips on checking ICs.
In reviewing the build in my head I realised that I have possibly made a fatal error.
Right at the end of the build, I realised that I swapped the position of a couple of headers, the power headers with the board-to-board header. I de-soldered them, but both boards had their ICs on. I removed them all before soldering the headers again./ I used a pump and the wick to clear the connection, I believe I did not damage the PCB, and there was no loss of trace in the pinholes after removing the header and the solder. How easily the ICs ca fry or short when submitted to what is technically short, without active current fed to the circuit, unless the soldering iron can discharge some?

TomWhitwell, 2023-03-17:
Hi, one way to check if the problem is the ICs is to swap the two 4081 chips. If the problem changes, then one of the chips is probably faulty and you should source some replacements (you can ask wherever you bought the kit). If the problem doesn't change, it's probably not a faulty chip - but might be a short or missing connection somewhere else 

Did you connect the power to the board? If not, you shouldn't have damaged the ICs - the tip of an iron shouldn't have any current.

drcat67, 2023-03-19:
HI
I swapped the chips, no change.
I would be worth changing the headers that I soldered in the wrong position originally as I have reused the same..although I did check each point with the meter after re-soldering and they checked ok, all other points I checked visually over and over..I don’t see anything wrong with it..loosing hope at this point.

drcat67, 2023-03-23:
Hi Tom, any further suggestions as to what I can check? perhaps an indication of point-to-point measurements with the tester? Also, I am having a problem tracing the appropriate headers (6pins male and female) There are quite a few formats some of them have a difference in the distance between pins and pin size. which are the specifics you know of?
Thanks

drcat67, 2023-03-24:
[image]
[image]
[image]
[image]

drcat67, 2023-04-05:
> <20230324_192124-0.jpg><20230324_192140-0.jpg><20230324_192153-0.jpg><20230324_192204-0.jpg>

Hi Tom,
So, after taking the pix of the two board, I left them in a container waiting to get some indications from you. After a few days worried about damaging them I put them back together and screw the machine back into the case. This morning I plugged it back in thinking that I should have least test if what I had working before was still ok and to my amusement the pulses were working. I believe at this point may theory of the re-used pins can be validated, there must be a dog pin somewhere.
Question now its? Does the pulses expander follow certain rule or is totally random?

Thanks

Luca