TomWhitwell/TuringMachine #199: 2 Turing machine mk2 both just built malfunctioning in the exact same way

pteagan, 2023-04-02:
I will do my best to explain this. I built 2 turing machine mk2 this week. I have built these in the past with zero issues..  Both are malfunctioning in the same way. They both seem to work except for the fact that they only create random drum like audio output without pitch change... so here is the story. I built the 1st one. It malfunctioned as described. I go over the whole build with a fine tooth comb and find nothing obvious. I contact Thonk. I end up ordering another complete kit and ask to buy a spare IC/transistor bag thinking maybe I had a bad chip or transistor on build 1. so I proceed to replace all the IC's and the transistors.. on build 1 .I have a hakko so it wasn't  messy. reassemble test and it does the same exact thing.... so I wait a day and I decide to build the 2nd complete kit I ordered. so I build this one with fine precision and detail as always but a little extra... so I finish assemble test and it does the exact same thing as build 1. so this morning I took appart build 2 and went over the entire build component by component even with a jewelers loop. I cannot find anything.. I have a short YouTube video here of what the modules are doing... so this is just a clock in and audio out.
https://youtu.be/Hy62vccgc5E
Any ideas...

TomWhitwell, 2023-04-02:
Just to check - there is no audio out on Turing - do you mean CV out is not changing?

pteagan, 2023-04-02:
Hello. There is audio out put. it sounds like a random drum sound without pitch changes. it is connected to a know good clock and a good known audio cable. see this video.https://youtu.be/Hy62vccgc5Eboth modules exhibit the same behavior.Sent from Samsung Galaxy smartphone.

Thonk-Steve, 2023-04-02:
Try patching the Turing output into the 1v/oct input of an oscillator, it's
a CV output, not an audio output.

pteagan, 2023-04-13:
Thanks. I embarrassingly realized my mistake. I should have remembered this from my past builds of the turning machine. I hope I'm not losing my memory already...

closed 2023-04-13 by pteagan