TomWhitwell/TuringMachine #200: Very Narrow Sweet point

drcat67, 2023-04-19:
Hi Tom,
After magically solving the problem with Pulses TM operated ok for a few days, but the sweet spot of randomness has slowly become very narrow. Let's say before I could go from 10 to 3 where 10 had some random 12 was totally random 3 very little random and the two locks L and R.  Now I barely move from 12 and is already locking a pattern.

Any Idea?
Thanks

drcat67, 2023-04-24:
@TomWhitwell, any Idea as to what it could be.? Thanks

Thonk-Steve, 2023-04-24:
Hi, have you tried re-calibrating?

drcat67, 2023-04-24:
Yes, calibration was still ok but even during calibration it is very noticeable how na=rrow the sweet spot has become. It glitches very marginally and then becomes a solid wave again

Thonk-Steve, 2023-04-24:
Try dialing the calibration back so the lock still holds at 'normal' clock
tempos, even if its not locking 100% using the audio rate calibration
method you can still get it to lock reliably.

drcat67, 2023-04-25:
Hi Steve,
I should have not touched it before changing that trim pot for calibrating.
The very first time I calibrated the module the plastic screw fell off the Tim leaving just a bit that I was able to carefully turn and calibrate.
I rotated grabbing on the little plastic left and the module went into all lights on, no sound nothing.
Switched off convinced that I had shorted it by touching the metal portion of the trim. Switched on twice and twice all red lights on. After an hour it was back to playing but so much worse, Random yes, but not nice random, so I don’t know For sure I am not experienced with putting together a module, but boy the amount of time I spent trying to make this work.
Could I have the specs for that trim pot please? I will change its and try to recalibrate.
Thanks

drcat67, 2023-04-26:
> Hi,
>
> As per the public BOM it's a 3362P trimmer at a 1M value, so this one would
> be fine -
> https://www.mouser.co.uk/ProductDetail/Bourns/3362P-1-105LF?qs=P7FeUmqzfS9tgrW%2F9VNbFg%3D%3D
> where are you located?
>
> Steve

Thanks you Steve,
Will see what I can do, not easy to find a business here or a shop that you can go and buy a 50cent component. Mouser and others have it for sure but they charge you from a minimum of $15 to $25 for delivery. Would you believe it?
I will ask the Australian Wigglers page on FB.

Cheers

Thonk-Steve, 2023-04-27:
One other option: We'd be happy to give you a free one on a future Thonk
order. I'm sure there are suppliers in Aus for this though, it's a very
standard part.

drcat67, 2023-04-28:
Hi Tom,
Thanks for this. I am sure there are, but you have never experienced what is like looking for a specific product in AU well, you are in for a surprise especially online. Australia has one of the most underdeveloped Ecommerce of any modern country, as a result searches online return a staggering amount of matches from either China or the USA and sprinkled in between some AU results.
I
Thanks again.

drcat67, 2023-04-28:
Hey Steve,
Laurie from Elby designs is going to send me one from his stock.