TomWhitwell/TuringMachine #201: TM2 locks with all lights steady and bright after a minute or so

alphonsom, 2023-04-23:
Newly built Turing Machine behaves for a minute or less, then all the LEDs light up fully bright and steady, and output ceases. It is calibrated as far as is possible within the time allowed before lockup. Even in the locked state, noise output is normal, and so is pulse output. Any hints or direction about where to look would be appreciated.

Thonk-Steve, 2023-04-23:
That's an unusual one. What do you have to do to get it working again?

alphonsom, 2023-04-23:
Switch off the power, and switch it back on again. Then the TM starts with all LEDs bright for a few seconds, followed by them going off in sequence from left to right in time with the incoming clock. Then there is a pause where all is dark for a few seconds. Then the normal behaviour starts for a while, and we go back to the beginning.

Thonk-Steve, 2023-04-23:
Hey, what are you clocking it with?

alphonsom, 2023-04-23:
A square wave out of an AI 011 VCO for calibration, or a clock signal from a Frequency Central Wave Runner LFO for normal. Same thing happens with either.

alphonsom, 2023-04-24:
There is no effect on any other module. I tried a different power supply and cables just to be sure, and the behaviour is the same. Could something be causing a draining of randomness on the shift register return switch? Or a gradual inversion failure?

Thonk-Steve, 2023-04-24:
Hi, did you build the module yourself? Did you use the IC pack from a Thonk
kit?

alphonsom, 2023-04-24:
Yes to both questions.

Thonk-Steve, 2023-04-24:
Ok the best step next would be to remove the front panel and post good
photos of both sides of both PCBs.