TomWhitwell/TuringMachine #202: Not truely "random"

rafael-azevedo, 2023-07-11:
I was able to calibrate the turing machine so the locking feature works. The issue I am having is that I always get a similar sequence a small ramp up from 0 in steps and then it holds the highest value set by the trim pot for a while and repeats. Now it does this randomly , but I never get step down or anything of that nature.

rafael-azevedo, 2023-07-11:
Adding two videos of it working. In this one with it in the unlocked position you can see if always hit the same voltages https://youtube.com/shorts/sOzEx5lxdU0?feature=share

a video of it with a "scope" view 

https://youtu.be/H7w2LE5eGns

rafael-azevedo, 2023-07-11:
posted pictures here of the build https://modwiggler.com/forum/viewtopic.php?p=3995928#p3995928

TomWhitwell, 2023-07-14:
The pictures look OK, was this built from a Thonk kit? 

One think to try would be to swap the two 4015 shift register chips and see if the behaviour changes

rafael-azevedo, 2023-07-14:
I am pretty sure its a thonk build. I picked it up second hand. It was working as expected before shipping. I swapped the two chips and the behavior stayed the same.

Thonk-Steve, 2023-07-14:
Hi Rafael,

I think you should return it and get a refund or insist that the seller
repairs it for you.

Regards

Steve
Thonk

rafael-azevedo, 2023-07-14:
Ya I worked out a deal with the seller already. Just trying to see if I can troubleshoot whats up. I ordered new IC chips if those don't work I'll just do a new build haha

Thonk-Steve, 2023-07-14:
Ok glad to hear that, there are some sellers who seem to think a buyer
should inherit their build problems :)

rafael-azevedo, 2023-07-17:
Any other recommendations on troubleshooting before I give up. I've replaced all the ICs outside of the DAC and the noise trimer pot and its still exhibiting the same behavior

rafael-azevedo, 2023-07-19:
Seems I have isolated the issue. The voltage out is not matching the bits in the shift register. If I understand correctly they should be multiples of 0.033v but I am getting 0.3V with 1 bit in the first position video attached explaining 

https://youtube.com/shorts/DnVebaOAveQ?feature=share

TomWhitwell, 2023-07-19:
Maybe something wrong around the resistors in the op amp that scales the output? Maybe if R26 was 50k or 500k you'd see this behaviour? Just a guess, I've never seen this before, and it would only explain incorrect voltage output, not incorrect sequences of bits.

rafael-azevedo, 2023-07-28:
Yup looks like there was an incorrect resistor for R26 I replaced it and had to jump it to p1 of the TC074. Now I am getting a new behavior where sequences longer then 5 don't work as expected https://youtube.com/shorts/4pj9n4bUo6s?si=-Ana2UskU1RuRxdt