TomWhitwell/TuringMachine #203: Unable to calibrate. Steady tone until 12 o clock

kengc, 2023-10-09:
This is a full thonk kit (plastic bag version). 

1) I have my TM in calibration 'mode' (following the settings from the TM calibration video)...

As mentioned i am unable to calibrate my TM. Turning the calibration trim pot does nothing (sound doesn't change). 
Turning the random knob from 5 o'clock up to 12 o' clock and the calibration sound cuts out. 

Here a video showing this behaviour (will need to turn up your sound for this video to hear the calibration tone)

[https://youtube.com/shorts/V4v92kzYj6o?feature=share](url)

2) NOT in calibration mode. Just sending a clock (100 bpm) to the TM and have the random knob parked at 12 o' clock. I click down on the switch to remove some bits and we can see that the TM doesn't really generate randomness and, in fact, toward the end of the following video all LEDs eventually are lit and no changes happen after this. 

[https://youtube.com/shorts/51hkEWINENg?feature=share](url)

Some trouble shooting steps i have taken:
1) swapped the two 4015 chips places
2) inspect soldering work and touched up as precaution
3) doubled checked all component position/placements against build doc
4) google search, including this help git for TM, and not seeing anything so far that helps my situation

I feel like one of the shift reg chips might be buggered or perhaps the big random knob is wonky? Any suggestions would be greatly appreciated. 

[image]
[image]
[image]

Thonk-Steve, 2023-10-10:
Hi Kevin,

Thanks for the info, despite the calibration not seeming to work, is the large knob having any effect on the output when operating at 'normal' tempo like in your second video?

Also is the white noise output working and delivering a loud signal?

Steve

OzzyMantis, 2023-12-03:
I am also having the almost exact problem as you. Has there been any resolution or any advice in troubleshooting the issues?

kengc, 2023-12-04:
Unfortunately my unit is still doing this

kengc, 2023-12-27:
Ok I solved my problem. Another clue i had forgotten about was my module was sending some pulses out of the white noise jack. Not the same pulses as the dedicated pulses jack, but similar. Anyway after following all of the white noise traces on the board i suspected something had to be up with the Link 1 header. 

Upon very close inspection with a bright LED light and my magnify glasses i could see a shiny reflection that appeared to go from pin 1 (white noise) to pin 2 (step 16) check photo with yellow indication line: 

[image]

Some might not be aware but one can gently lift the plastic shroud up off the circuit board for these headers to better inspect the pins (i used the smallest flat head screwdriver from my eye glasses kit, it's quite small), sure enough a loose piece of solder was there connecting the aforementioned pins. After removing the solder, and gently pushing the plastic shroud back down, the module works exactly how it should. At 12o'clock it's randomizing like a champ. No more pulses out of white noise and i could finally calibrate it properly.

closed 2023-12-27 by kengc