TomWhitwell/TuringMachine #209: Calibration Help - Only Hearing Clock Pulses

glassofJAMESon, 2024-01-10:
Hey Everyone,
Just built my first TM and when I went to calibrate it I I'm only getting clock pulses / noise -- no tones of any kind. Has anyone heard anything like this before? 

I'm sure it's some sort of user error -- but I have no idea where to begin. 
Thank you so much!

[IMG_0921.MOV.zip](https://github.com/TomWhitwell/TuringMachine/files/13883864/IMG_0921.MOV.zip)

TomWhitwell, 2024-01-10:
Hi, Haven't been able to watch the video, but the output is CV, not audio - pass it into an oscillator  voltage input to hear the tones.  I don't know if that answers the question.

glassofJAMESon, 2024-01-16:
> o

Hey Tom,
Thank you so much for the response! The above issue was user error - I didn't have the panel on correctly so was using the wrong output (bonehead move) 

After I connected everything, it seems to work! It produces random voltages, but it does not lock when turned clock/counter clockwise.

When calibrating using the vimeo video, I can't seem to find the stable tone on the trimmer. I get glitches from clockwise to counter-clockwise. 

Again, I'm sure this may be another user error - but do you have any thoughts? 

THANK YOU SO MUCH TOM! 

[image]
[image]